lnis-uofu/SOFA #97: ICC2 scripts?

nachiket, 2021-02-09:
Hi -- would you consider open-sourcing the ICC2 scripts? I'm interested to see how the backend place-and-route flow works.

ganeshgore, 2021-02-09:
Hello, Due to legal reasons, We can not share ICC2 scripts publicly. 
We need to have a three way NDA with SNPS, even for sharing scripts with other parties. 
But we have a publication coming up in ISPD2021, 
which will give a fair idea of how you can quickly place-and-route the generated netlist. 

Also, We have a short term plan to implement similar place-and-route flow in opensource EDA tools, 
which should be ready before Summer 2021.

pegaillardon, 2021-02-09:
We also have in our pipeline to implement a "wrapper" that can hide the proprietary commands (from the EDA vendors) to make things easier to exchange. It should take a few months.