lnis-uofu/SOFA #124: write_verilog_testbench is not defined in latest OpenFPGA release

abdullahyildiz, 2021-06-29:
I get the following error when I try to run `make runOpenFPGA` with the latest OpenFPGA release:

```
Command line to execute: write_verilog_testbench --file ./SRC         --reference_benchmark_file_path counter_output_verilog.v         --print_top_testbench         --print_preconfig_top_testbench       \
  --print_simulation_ini ./SimulationDeck/simulation_deck.ini         --explicit_port_mapping
Try to call a command 'write_verilog_testbench' which is not defined!
Fatal error occurred!
OpenFPGA Abort
```

It seems that the command write_verilog_testbench is [deprecated](https://github.com/lnis-uofu/SOFA/pull/123).

What is the latest compatible OpenFPGA commit id to be used within SOFA project?

tangxifan, 2021-06-29:
@abdullahyildiz You can find latest commit here: https://github.com/lnis-uofu/OpenFPGA/commit/ef3cbe708c9fd516500c09c22d97f1c6ee07d67d

abdullahyildiz, 2021-06-30:
When I try to run make runOpenFPGA with the OpenFPGA release https://github.com/lnis-uofu/OpenFPGA/commit/ef3cbe708c9fd516500c09c22d97f1c6ee07d67d, I get `write_full_testbench is not defined error` within openfpgashell.log:

[openfpgashell.log](https://github.com/lnis-uofu/SOFA/files/6738731/openfpgashell.log)

tangxifan, 2021-07-01:
@abdullahyildiz Sorry for the inconvenience. I merged the pull request https://github.com/lnis-uofu/SOFA/pull/123

Now you should be able to use the latest OpenFPGA on master branch.

abdullahyildiz, 2021-07-02:
@tangxifan,

No problem. Thanks also for your support.

This time I get the following error:

```
 Write autocheck testbench for FPGA top-level Verilog netlist for 'counter'
Will use 78766 configuration clock cycles to top testbench
# Write autocheck testbench for FPGA top-level Verilog netlist for 'counter' took 0.00 seconds (max_rss 59.8 MiB, delta_rss +0.0 MiB)
Write Verilog full testbenches for FPGA fabric
 took 0.00 seconds (max_rss 59.8 MiB, delta_rss +0.0 MiB)

Command line to execute: --reference_benchmark_file_path counter_output_verilog.v         --explicit_port_mapping
Try to call a command '--reference_benchmark_file_path' which is not defined!
Fatal error occurred!
OpenFPGA Abort

Finish execution with 1 errors

The entire OpenFPGA flow took 1.46976 seconds

Thank you for using OpenFPGA!
1
```

Is something wrong with my setup?

abdullahyildiz, 2021-07-24:
@tangxifan,

I think the problem arises from a typo in L58 in [generate_testbench.openfpga](https://github.com/lnis-uofu/SOFA/blob/45184c35ea578daecef76b26557b25fdbf6e8edd/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/generate_testbench.openfpga): https://github.com/lnis-uofu/SOFA/blob/45184c35ea578daecef76b26557b25fdbf6e8edd/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/generate_testbench.openfpga#L58

There should be a backslash at the end of the line.

tangxifan, 2021-07-24:
@abdullahyildiz Thanks for pointing this out. Should fix it asap

closed 2021-08-17 by abdullahyildiz