lnis-uofu/SOFA #126: Functional Simulaton of SOFA RTL Design

abdullahyildiz, 2021-08-18:
I did the following in order to run functional simulation of FPGA1212_SOFA_HD design with `and` benchmark in Icarus Verilog:

```bash
# STEP 1 ################################################
> mkdir $OPENFPGA_PATH/openfpga_flow/tasks/FPGA22_HIER_SKY_PNR/
#########################################################

# STEP 2 ################################################
# create the following hierarchy:
# $OPENFPGA_PATH/openfpga_flow/tasks/FPGA22_HIER_SKY_PNR/sc_verilog
# ├── digital_io_hd.v
# └── fpga_top.v
# $OPENFPGA_PATH/openfpga_flow/tasks/FPGA22_HIER_SKY_PNR/skywater
# └── libraries
#     └── sky130_fd_sc_hd -> $OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd
#########################################################

# STEP 3 ################################################
> diff digital_io_hd.v digital_io_hd.v~
4c4
< `include "sky130_fd_sc_hd__ebufn_4.v"
---
>
#########################################################

# STEP 4 ################################################
> cd FPGA1212_SOFA_HD_PNR
#########################################################

# STEP 5 ################################################
> git diff FPGA1212_SOFA_HD_task/generate_fabric.openfpga
diff --git a/FPGA1212_SOFA_HD_PNR/FPGA1212_SOFA_HD_task/generate_fabric.openfpga b/FPGA1212_SOFA_HD_PNR/FPGA1212_SOFA_HD_task/generate_fabric.openfpga
index 8e9a0a6..227cc84 100644
--- a/FPGA1212_SOFA_HD_PNR/FPGA1212_SOFA_HD_task/generate_fabric.openfpga
+++ b/FPGA1212_SOFA_HD_PNR/FPGA1212_SOFA_HD_task/generate_fabric.openfpga
@@ -42,13 +42,10 @@ write_fabric_verilog \
     --include_timing \
     --verbose
 
-write_verilog_testbench \
-    --file ./SRC \
-    --reference_benchmark_file_path ${REFERENCE_VERILOG_TESTBENCH} \
-    --print_top_testbench \
-    --print_preconfig_top_testbench \
-    --print_simulation_ini ./SimulationDeck/simulation_deck.ini \
-    --explicit_port_mapping
+write_full_testbench --file ./SRC \
+        --bitstream fabric_bitstream.bit \
+        --reference_benchmark_file_path ${REFERENCE_VERILOG_TESTBENCH} \
+        --explicit_port_mapping
 
 # Write the SDC files for PnR backend
 #  - Turn on every options here

#########################################################

# STEP 6 ################################################
> make clean runOpenFPGA TASK_FILENAME=task_generation.conf
#########################################################

# STEP 7 ################################################
> cd FPGA1212_SOFA_HD_task/run001/vpr_arch/top/MIN_ROUTE_CHAN_WIDTH
#########################################################

# STEP 8 ################################################
> diff -u SRC/top_autocheck_top_tb.v SRC/top_autocheck_top_tb.v.bak
--- SRC/top_autocheck_top_tb.v	2021-08-18 15:54:38.507455822 +0300
+++ SRC/top_autocheck_top_tb.v.bak	2021-08-18 15:54:26.407275642 +0300
@@ -43,13 +43,13 @@
 	reg [0:0] b;
 
 // ----- FPGA fabric outputs -------
-	wire [0:0] out_c_fpga;
+	wire [0:0] out:c_fpga;
 
 // ----- Benchmark outputs -------
-	wire [0:0] out_c_benchmark;
+	wire [0:0] out:c_benchmark;
 
 // ----- Output vectors checking flags -------
-	reg [0:0] out_c_flag;
+	reg [0:0] out:c_flag;
 
 // ----- Error counter: Deposit an error for config_done signal is not raised at the beginning -----
 	integer nb_error= 1;
@@ -153,8 +153,8 @@
 // ----- Blif Benchmark input b is mapped to FPGA IOPAD gfpga_pad_EMBEDDED_IO_HD_SOC_IN[65] -----
 	assign gfpga_pad_EMBEDDED_IO_HD_SOC_IN[65] = b[0];
 
-// ----- Blif Benchmark output out_c is mapped to FPGA IOPAD gfpga_pad_EMBEDDED_IO_HD_SOC_OUT[63] -----
-	assign out_c_fpga[0] = gfpga_pad_EMBEDDED_IO_HD_SOC_OUT[63];
+// ----- Blif Benchmark output out:c is mapped to FPGA IOPAD gfpga_pad_EMBEDDED_IO_HD_SOC_OUT[63] -----
+	assign out:c_fpga[0] = gfpga_pad_EMBEDDED_IO_HD_SOC_OUT[63];
 
 // ----- Wire unused FPGA I/Os to constants -----
 	assign gfpga_pad_EMBEDDED_IO_HD_SOC_IN[0] = 1'b0;
@@ -305,7 +305,7 @@
 	top REF_DUT(
 		.a(a),
 		.b(b),
-		.c(out_c_benchmark)	);
+		.c(out:c_benchmark)	);
 // ----- End reference Benchmark Instanication -------
 
 // ----- Begin bitstream loading during configuration phase -----
@@ -349,7 +349,7 @@
 		a <= 1'b0;
 		b <= 1'b0;
 
-		out_c_flag[0] <= 1'b0;
+		out:c_flag[0] <= 1'b0;
 	end
 
 // ----- Input Stimulus -------
@@ -366,18 +366,18 @@
 		if (1'b1 == sim_start[0]) begin
 			sim_start[0] <= ~sim_start[0];
 		end else begin
-			if(!(out_c_fpga === out_c_benchmark) && !(out_c_benchmark === 1'bx)) begin
-				out_c_flag <= 1'b1;
+			if(!(out:c_fpga === out:c_benchmark) && !(out:c_benchmark === 1'bx)) begin
+				out:c_flag <= 1'b1;
 			end else begin
-				out_c_flag<= 1'b0;
+				out:c_flag<= 1'b0;
 			end
 		end
 	end
 
-	always@(posedge out_c_flag) begin
-		if(out_c_flag) begin
+	always@(posedge out:c_flag) begin
+		if(out:c_flag) begin
 			nb_error = nb_error + 1;
-			$display("Mismatch on out_c_fpga at time = %t", $realtime);
+			$display("Mismatch on out:c_fpga at time = %t", $realtime);
 		end
 	end

#########################################################

# STEP 9 ################################################
> iverilog -o sim SRC/top_include_netlists.v -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/inv -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/buf -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/or2 -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/mux2 -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/sdfxtp -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/dfxtp -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/or2b -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/ebufn
#########################################################

# STEP 10 ###############################################
> vvp sim
#########################################################
```

However, out_c_fpga is displayed as 'X':

[image]

Is the problem something related to Icarus Verilog?

abdullahyildiz, 2021-08-21:
I used the functional FF model instead of behavioral one to fix the problem:

```bash
> iverilog -o sim -DFUNCTIONAL -DUNIT_DELAY=#0 SRC/top_include_netlists.v -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/inv -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/buf -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/or2 -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/mux2 -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/sdfxtp -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/dfxtp -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/or2b -I$OPENLANE_ROOT/pdks/skywater-pdk/libraries/sky130_fd_sc_hd/latest/cells/ebufn
```

tangxifan, 2021-08-22:
@abdullahyildiz You are right. We used the functional FF model in our verification. We fixed the same bug as you did. I encourage you to report the issue to the maintainers of the skywater open pdk. Just like we did before: https://github.com/google/skywater-pdk/issues/197

closed 2021-08-23 by abdullahyildiz