lnis-uofu/SOFA #129: Failed to execute openfpga flow

msaideroglu, 2021-09-22:
I'm getting this error during first test:
"ERROR (00_counter_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_counter_MIN_ROUTE_CHAN_WIDTH
.
.
.'python3 /home/msaid/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py ... _task/run001 /vpr_arch/counter /MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port' returned non-zero exit status 0.
X X X X X X Failed to generate netlist X X X X X X
"
I have a running OpenFPGA installation.

ganeshgore, 2021-09-22:
Can you share the `openfpgashell.log` file from the run directory

msaideroglu, 2021-09-22:
Thanks for fast return:
Log file:
* * * * * * * * * * * * * * * * * * * * 
RunDirectory : /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001
python3 /home/msaid/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/vpr_arch.xml /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/BENCHMARK/counter/counter.v --top_module counter --run_dir /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/generate_testbench.openfpga --openfpga_arch_file /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/openfpga_arch.xml --openfpga_sim_setting_file /home/msaid/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --external_fabric_key_file /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/fabric_key.xml --openfpga_vpr_device_layout 12x12 --openfpga_vpr_route_chan_width 60 --power --power_tech /home/msaid/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --arch_variable_file /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/design_variables.yml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port
* * * * * * * * * * * * * * * * * * * * 
INFO - Validating commnad line arguments
INFO - Run directory : /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH
INFO - Running "yosys_vpr" Flow
INFO - Extracted lut_size size from arch XML = 4
INFO - Running Yosys with lut_size = 4
INFO - Launching Run yosys 
INFO - Run yosys is written in file yosys_output.log
INFO - ACE2 output is written in file counter_ace2_output.txt
INFO - blif_3args output is written in file counter_blif_3args_output.txt
INFO - Launching Yosys 
INFO - Yosys is written in file yosys_rewrite.log
INFO - Runing OpenFPGA Shell Engine 
INFO - Launching OpenFPGA Shell Run 
INFO - OpenFPGAShell Revision: a2b2642e-dirty Compiled: 2021-05-21T02:43:13
ERROR - OpenFPGA Shell Run run failed with returncode 1
ERROR - command /home/msaid/OpenFPGA/openfpga/openfpga -batch -f counter_run.openfpga
ERROR - Current working directory : /home/msaid/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH
ERROR - Failed to run OpenFPGA Shell Run task
ERROR - Exiting . . . . . .

msaideroglu, 2021-09-22:
[openfpgashell.log](https://github.com/lnis-uofu/SOFA/files/7213523/openfpgashell.log)

msaideroglu, 2021-09-22:
[openfpgashell.log](https://github.com/lnis-uofu/SOFA/files/7213527/openfpgashell.log)

msaideroglu, 2021-09-22:
[00_counter_MIN_ROUTE_CHAN_WIDTH_out.log](https://github.com/lnis-uofu/SOFA/files/7213669/00_counter_MIN_ROUTE_CHAN_WIDTH_out.log)

ganeshgore, 2021-09-23:
[image]

`write_full_testbench` command is deprecated, please replace that command in your `*.openfpga` file with  following 
`write_preconfigured_testbench --file ./SRC --reference_benchmark_file_path ${REFERENCE_VERILOG_TESTBENCH} --explicit_port_mapping`