lnis-uofu/SOFA #130: is skywater130 pdk supported by synopsys

msaideroglu, 2021-09-27:
Hi. How did you do PnR flow with this pdk on ICC2? are there necessary files in sky130a directory, if not where should I obtain it? from synopsys etc.

ganeshgore, 2021-09-27:
You can find SKY130 files to use with Synopsys ICC2 in the following repository.
https://github.com/ganeshgore/skywater-pdk/tree/ganesh_dev/vendor/synopsys

Hope that helps

msaideroglu, 2021-09-28:
Thank you very much. Then this files can be used on DC for synthesizing as well?

ganeshgore, 2021-09-28:
Yes, you should be able to use it with DC as well.

msaideroglu, 2021-09-29:
Thank you very much

msaideroglu, 2021-11-20:
lc.tcl and lm.tcl. includes:
"source ./setup.tcl
source ${ROOT}/scripts/util.tcl"
However there is skywater130_setup.tcl file in directory and it also includes:
"set DESIGN_REF_PATH "/slowfs/cae678/diegob/testing/powercae_scripts/libraries/google-skywater/results""
another thing that is included in setup.tcl is 
"set MIN_ROUTING_LAYER            "M2"   ;# Min routing layer
set MAX_ROUTING_LAYER            "M8"   ;# Max routing layer"
I thought tha max routing layer is M5. Furthermore, source ${ROOT}/scripts/util.tcl should be source ./util.tcl I think. Are those scripts ready to run on Synopsys machine? If not could you upload updated scripts please? Because they are quite useful.

ganeshgore, 2021-11-24:
Hello @msaideroglu this question is regarding `ganeshgore/skywater-pdk` repo right

ganeshgore, 2021-11-24:
I added `make icc2_console` recipe to makefile.

msaideroglu, 2021-11-25:
Thanks for update. However it gives error during lm.tcl sourcing:
the error is 
"foreach db_file [glob -directory ${db_dir} *.db] {
    read_db ${db_file}
}
Error: no files matched glob pattern "*.db"
	Use error_info for more info. (CMD-013)
Information: script '/home/msaid/skywater-pdk/vendor/synopsys/scripts/lm.tcl'
	stopped at line 32 due to error. (CMD-081)
Extended error info:
no files matched glob pattern "*.db"
    while executing
"glob -directory ${db_dir} *.db"
    (file "/home/msaid/skywater-pdk/vendor/synopsys/scripts/lm.tcl" line 32)
 -- End Extended Error Info
lm_shell>
"

please look at log files:
[logs.tar.gz](https://github.com/lnis-uofu/SOFA/files/7603053/logs.tar.gz)

msaideroglu, 2021-11-28:
Hi @ganeshgore. I have some questions. When I clone ganeshgore/skywater-pdk repo in "https://github.com/ganeshgore/skywater-pdk/tree/ganesh_dev/vendor/synopsys", it gives pip dependency error during "make  sky130_fd_sc_hd". However if I clone skywater-pdk master repo then it runs "make  sky130_fd_sc_hd" successfully and builds the pdk. My question is that can we run Synopsys make scripts in vendor/synopsys in skywater-pdk master repo or it needs ganeshgore/skywater-pdk repo anyway? Because when I try to run the synopsys scripts in skywater-pdk master repo it gives some errors explained in previous comment.

msaideroglu, 2021-11-29:
source ./setup.tcl should have not been replaced in lc.tcl and mw.tcl. When I added the command again it runned successfully setup_pnr.tcl. Thanks @ganeshgore

closed 2022-04-18 by mattvlsi