lnis-uofu/SOFA #132: Some question

kangliyu1, 2021-12-06:
How can I change the Skywater 130nm PDK library of the underlying module to a module designed by myself

ganeshgore, 2021-12-06:
> 1、How can I change the Skywater 130nm PDK library of the underlying module to a module designed by myself

If you plan to change the PDK you need to perform PnR again (ref and to Q3), if you are still using Skywater 130 you can reuse some of the base modules (CB, SB, CLB) shared in this repo.

> 2、Is this a 12*12 CLB array? Only 1 LUT per CLB module

This is K4N8 architure (8LUTs perCLB)

> 3、How to expand the scale of the design

You can expand the size of FPGA for architecture evaluation in the XML file, but for GDS of the scaled design, you need to perform place and route yourself. This repository does not share PnR scripts as we used commercial tools to perform PnR

kangliyu1, 2021-12-07:
> If you plan to change the PDK you need to perform PnR again (ref and to Q3), if you are still using Skywater 130 you can reuse some of the base modules (CB, SB, CLB) shared in this repo.
>
>
> This is [...]

It seems that the underlying modules of Skywater 130nm PDK cannot be synthesis. How do you synthesis them?