lnis-uofu/SOFA #137: FPGA1212_QLSOFA_HD_task reproducing fabric netlists

mattvlsi, 2022-04-04:
When I run **make clean runOpenFPGA TASK_FILENAME=task_generation.conf** to reproduce same Verilog netlists that exists in repo and to run some tasks on it, it executes to code but there are some missing verilog files in /SRC.

**counter_include_netlists.v** contents are:

```
`include "./SRC/fabric_netlists.v"

`include "counter_output_verilog.v"

`include "./SRC/counter_autocheck_top_tb.v"
```

However there is no fabric_netlists.v file in /SRC.

How can I solve this issue?

Thanks.

@ganeshgore

ganeshgore, 2022-04-04:
Seems link you are not linking the design variable file. There is a ${OUTPAD_DELAY} in the architecture file which is not resolved.

This is where we define the design variable file  
https://github.com/lnis-uofu/SOFA/blob/e508bdd9056639101993f84a215ab10354659ad6/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/config/task_generation.conf#L16

The variable is defined here 
https://github.com/lnis-uofu/SOFA/blob/e508bdd9056639101993f84a215ab10354659ad6/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/design_variables.yml#L12

ganeshgore, 2022-04-04:
Some file path is missing.

mattvlsi, 2022-04-06:
Thanks @ganeshgore. I successfully obtain fabric netlists now. I'm running **generate_tetbench.openfpga** with **task_simulation.conf** in task directory. Should I expect to see iverilog.txt and and2_formal.vcd files in task directory just like in generate_testbench tasks in OpenFPGA? Because it isn't generated. Or how can I do the iverilog simulations by manually?

closed 2022-04-18 by mattvlsi