lnis-uofu/SOFA #140: Instructions for synthesis and PnR for SOFA?

gussmith23, 2022-11-07:
Hi!

I'm pretty lost on how to do synthesis and PnR for SOFA.

So far I have:
1. Downloaded and built OpenFPGA.
2. Run a few test commands, e.g.
   ```
   python OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py --flow_config SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/config/task_benchmarks.conf --arch_file SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/vpr_arch.xml --benchmark_files SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/BENCHMARK/and2/and2.v --openfpga_arch_file SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/openfpga_arch.xml --openfpga_shell_template /home/gus/OpenFPGA/openfpga_flow/openfpga_shell_scripts/example_script.openfpga --top_module and2
   ```
   
 I've ended up getting errors like:
 ```
 ERROR - -->>Error 1: /home/gus/OpenFPGA/tmp/arch/vpr_arch.xml:76 Unexpected attribute 'capacity' found on node 'tile'. Expected (possibly) one of: 'name', 'width', 'height', or 'area'.
ERROR - -->>Error 2: Command 'vpr' execution has fatal errors
```

All I need is to be able to synthesize simple Verilog files---can someone help me out?

gussmith23, 2022-11-07:
I wondered if it was maybe an OpenFPGA versioning issue. If so, can someone let me know what version I should be using for SOFA?

ganeshgore, 2022-11-07:
Hey @gussmith23,
That's correct, please use this version of OpenFPGA
https://github.com/lnis-uofu/SOFA/blob/e508bdd9056639101993f84a215ab10354659ad6/FPGA1212_SOFA_HD_PNR/FPGA1212_SOFA_HD_Verilog/OpenFPGAEngine.info#L1-L4

gussmith23, 2022-11-08:
Thank you. I'm still stuck. What am I doing wrong here?

```
boba ➜  FPGA1212_QLSOFA_HD_PNR git:(master) pwd
/home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR
boba ➜  FPGA1212_QLSOFA_HD_PNR git:(master) OPENFPGA_PATH=~/OpenFPGA/ make
INFO (     MainThread) - Set up to run 2 Parallel threads
INFO (     MainThread) - Currently running task FPGA1212_QLSOFA_HD_task
ERROR (     MainThread) - Task directory [/home/gus/OpenFPGA/openfpga_flow/tasks/FPGA1212_QLSOFA_HD_task] not found
ERROR (     MainThread) - Exiting . . . . . .
INFO (     MainThread) - Set up to run 2 Parallel threads
INFO (     MainThread) - Currently running task FPGA1212_QLSOFA_HD_task
ERROR (     MainThread) - Task directory [/home/gus/OpenFPGA/openfpga_flow/tasks/FPGA1212_QLSOFA_HD_task] not found
ERROR (     MainThread) - Exiting . . . . . .
X X X X X X Failed to generate netlist X X X X X X
```

OpenFPGA doesn't seem to be able to handle tasks that aren't in its `tasks/` directory.

gussmith23, 2022-11-08:
I can manually set the `task_dir` in `/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_task.conf`, but then I get 
```
/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_task.conf
INFO (     MainThread) - Set up to run 2 Parallel threads
INFO (     MainThread) - Currently running task FPGA1212_QLSOFA_HD_task
INFO (     MainThread) - Task execution completed
INFO (     MainThread) - Set up to run 2 Parallel threads
INFO (     MainThread) - Currently running task FPGA1212_QLSOFA_HD_task
INFO (     MainThread) - Created "run001" directory for current task run
INFO (     MainThread) - Running "yosys_vpr" flow
INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
INFO (     MainThread) - Created total 1 jobs
/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py:403: DeprecationWarning: currentThread() is deprecated, use current_thread() instead
  thread_name = threading.currentThread().getName()
/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py:403: DeprecationWarning: getName() is deprecated, get the name attribute instead
  thread_name = threading.currentThread().getName()
ERROR (00_counter_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_counter_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 426, in run_single_script
    raise subprocess.CalledProcessError(0, " ".join(command))
subprocess.CalledProcessError: Command 'python3 /home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/vpr_arch.xml /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/BENCHMARK/counter/counter.v --top_module counter --run_dir /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/generate_testbench.openfpga --openfpga_arch_file /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/openfpga_arch.xml --openfpga_sim_setting_file /home/gus/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --external_fabric_key_file /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/fabric_key.xml --openfpga_vpr_device_layout 12x12 --openfpga_vpr_route_chan_width 60 --power --power_tech /home/gus/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port' returned non-zero exit status 0.
X X X X X X Failed to generate netlist X X X X X X
```

gussmith23, 2022-11-11:
Bumping on this! We're working on an ISCA submission so this is pretty urgent!

ganeshgore, 2022-11-11:
Can you share openfpgashell.log file

gussmith23, 2022-11-11:
That file doesn't seem to exist anywhere in the SOFA or OpenFPGA directory. I was able to get a little bit more detail by printing out the error that occurs on line https://github.com/lnis-uofu/OpenFPGA/blob/3a80af3408bcc76d5d5a5fe4def15b3da7682ac3/openfpga_flow/scripts/run_fpga_task.py#L414

```
INFO (     MainThread) - Set up to run 2 Parallel threads
INFO (     MainThread) - Currently running task FPGA1212_QLSOFA_HD_task
INFO (     MainThread) - Removing run_dir run001
INFO (     MainThread) - Removing run_dir latest
INFO (     MainThread) - Task execution completed
INFO (     MainThread) - Set up to run 2 Parallel threads
INFO (     MainThread) - Currently running task FPGA1212_QLSOFA_HD_task
INFO (     MainThread) - Created "run001" directory for current task run
INFO (     MainThread) - Running "yosys_vpr" flow
INFO (     MainThread) - Found 1 Architectures 1 Benchmarks & 1 Script Parameters
INFO (     MainThread) - Created total 1 jobs
/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py:402: DeprecationWarning: currentThread() is deprecated, use current_thread() instead
  thread_name = threading.currentThread().getName()
/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py:402: DeprecationWarning: getName() is deprecated, get the name attribute instead
  thread_name = threading.currentThread().getName()
INFO - Validating commnad line arguments
INFO - Run directory : /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 996, in <module>
    main()
  File "/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 247, in main
    prepare_run_directory(args.run_dir)
  File "/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py", line 414, in prepare_run_directory
    archfile.write(tmpl.substitute(script_env_vars["PATH"]))
  File "/home/gus/.pyenv/versions/3.10.6/lib/python3.10/string.py", line 121, in substitute
    return self.pattern.sub(convert, self.template)
  File "/home/gus/.pyenv/versions/3.10.6/lib/python3.10/string.py", line 114, in convert
    return str(mapping[named])
KeyError: 'L1_SB_MUX_DELAY'

ERROR (00_counter_MIN_ROUTE_CHAN_WIDTH) - Failed to execute openfpga flow - 00_counter_MIN_ROUTE_CHAN_WIDTH
Traceback (most recent call last):
  File "/home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_task.py", line 427, in run_single_script
    raise subprocess.CalledProcessError(0, " ".join(command))
subprocess.CalledProcessError: Command 'python3 /home/gus/OpenFPGA/openfpga_flow/scripts/run_fpga_flow.py /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/vpr_arch.xml /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/BENCHMARK/counter/counter.v --top_module counter --run_dir /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH --fpga_flow yosys_vpr --openfpga_shell_template /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/generate_testbench.openfpga --openfpga_arch_file /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/openfpga_arch.xml --openfpga_sim_setting_file /home/gus/OpenFPGA/openfpga_flow/openfpga_simulation_settings/auto_sim_openfpga.xml --external_fabric_key_file /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/arch/fabric_key.xml --openfpga_vpr_device_layout 12x12 --openfpga_vpr_route_chan_width 60 --power --power_tech /home/gus/OpenFPGA/openfpga_flow/tech/PTM_45nm/45nm.xml --vpr_fpga_verilog --vpr_fpga_verilog_dir /home/gus/SOFA/FPGA1212_QLSOFA_HD_PNR/FPGA1212_QLSOFA_HD_task/run001/vpr_arch/counter/MIN_ROUTE_CHAN_WIDTH --vpr_fpga_x2p_rename_illegal_port' returned non-zero exit status 0.
X X X X X X Failed to generate netlist X X X X X X
```

It seems like some config value `L1_SB_MUX_DELAY` isn't getting set?