lnis-uofu/SOFA #141: How did you harden tiles?

WazaAbdulkadir, 2022-11-21:
I want to implement OpenFPGA flow, create my own FPGA using an open-source backend tool (OpenLane), and send it to GFMPW0. 

As it is evaluated in the  [link](https://caravel-user-project.readthedocs.io/en/latest/) there are three options to harden in the Caravel environment: 

1- Hardening the user macro(s) first, then inserting it in the user project wrapper with no standard cells on the top level:

    If this is the right way the path in the backend tool might be like this I assume:
    Creating CLB, Switch Block, x and y connection macros, and then creating a macro that includes all of them. 
    Using the last macro 4 times and create 2*2 CLB. Lastly, making I/O connections.  (Does this seems right?)

2- Flattening the user macro(s) with the user_project_wrapper 
3- Placing multiple macros in the wrapper along with standard cells on the top level 

I want to know about: 

how you did harden it and what do you recommend for the OpenLane backend tool from the above options? 

Finally, is OpenFPGA compatible with GF-180 pdk?

tangxifan, 2022-11-22:
@WazaAbdulkadir 

- I do not see an issue for OpenFPGA to support the GF-180 pdk. OpenFPGA supports standard cells to build FPGA primitives. If your standard cell library contains the following type of gates, 
  - MUX2
  - Buffer
  - Inverter
  - D-type FF

you should be able to use OpenFPGA to obtain netlists.

- Your questions regarding physical design makes sense to me. Our practice is that you do PnR on each building blocks separately, such as CLB, CBx, CBy, SB, and then assemble them at the top-level. Note that you may need to add pre-routed network for clock and reset signals to each building blocks. See details: https://dl.acm.org/doi/abs/10.1145/3439706.3447047

- Regarding other questions about physical design, I suggest you to contact @ganeshgore who has enriched experience in the physical design using OpenFPGA. He has worked with me on several tape-out projects, including the Skywater MPWs. He knows how to make FPGA chips.

WazaAbdulkadir, 2022-11-23:
Thanks a lot for the clear answers. 

I have sent an e-mail to @ganeshgore asking the same questions. 
Would like to have an answer from him about physical design.

ganeshgore, 2022-11-24:
Hello @WazaAbdulkadir, sorry for the delayed reply I was busy with some other deadlines.

1. We did not use OpenLane to harden tiles in `SOFA` and `SOFA-Plus` designs; we used Synopsys ICC2. Although we have some students working on porting design flow to OpenLane but is taking longer than expected 
2. "**Hardening the user macro(s) first, ....**" that looks like the correct approach that will give you the shortest runtime and fastest iteration, but you need good design planning to achieve that
3. **Creating CBX and CBY macro makes sense**; in the case of homogeneous design using OpenFPGA, you will have anything between 8-15 unique modules
4. **Flattening the _user_project_wrapper_** and adding macros and standard cells in it seem correct
5. I believe you can achieve this design flow with OpenLane (I have yet to do it myself, but very interested in developing it). On the other hand, commercial tools will make it very easy to achieve this. The most important part of design planning.

**Is OpenFPGA compatible with GF-180 pdk?**
OpenFPGA has no dependency on which PDK is used. Everything else should go well if you specify the correct standard cells.

Let me know if you have any more questions

closed 2023-06-05 by WazaAbdulkadir