lnis-uofu/SOFA #143: Timing report details

msaideroglu, 2022-12-22:
I am looking at SOFA_HD task. design_variables.yml includes delay parameters of primitives. When I run a benchmark on SOFA_HD, for instance a counter benchmark the setup timing report says: 
`#Path 1
Startpoint: q[3].Q[0] (.latch clocked by clk)
Endpoint  : q[7].D[0] (.latch clocked by clk)
Path Type : setup

Point                                                             Incr      Path
--------------------------------------------------------------------------------
clock clk (rise edge)                                            0.000     0.000
clock source latency                                             0.000     0.000
clk.inpad[0] (.input)                                            0.000     0.000
q[3].clk[0] (.latch)                                             0.110     0.110
q[3].Q[0] (.latch) [clock-to-output]                             0.430     0.540
$abc$215$new_n25_.in[0] (.names)                                10.320    10.860
$abc$215$new_n25_.out[0] (.names)                                1.140    12.000
$abc$215$new_n30_.in[3] (.names)                                10.890    22.890
$abc$215$new_n30_.out[0] (.names)                                1.140    24.030
n30.in[2] (.names)                                               4.450    28.480
n30.out[0] (.names)                                              0.860    29.340
q[7].D[0] (.latch)                                               1.140    30.480
data arrival time                                                         30.480

clock clk (rise edge)                                            0.000     0.000
clock source latency                                             0.000     0.000
clk.inpad[0] (.input)                                            0.000     0.000
q[7].clk[0] (.latch)                                             0.110     0.110
clock uncertainty                                                0.000     0.110
cell setup time                                                 -0.390    -0.280
data required time                                                        -0.280
--------------------------------------------------------------------------------
data required time                                                        -0.280
data arrival time                                                        -30.480
--------------------------------------------------------------------------------
slack (VIOLATED)                                                         -30.760`
I can understand 0.11, 0.43, 1.14 and 0.86 ns etc. values comes directly from design variables. But I couldn't realize where those 10.32, 10.89, 4.45 ns values come from?

ganeshgore, 2023-01-09:
those are routing delays which are calculated based on RC value provided in the [design_variable.yml](https://github.com/lnis-uofu/SOFA/blob/master/FPGA1212_SOFA_HD_PNR/FPGA1212_SOFA_HD_task/design_variables.yml) file 
https://github.com/lnis-uofu/SOFA/blob/e508bdd9056639101993f84a215ab10354659ad6/FPGA1212_SOFA_HD_PNR/FPGA1212_SOFA_HD_task/design_variables.yml#L5-L10

These delays are dynamically calculated based on the routing wire length, hence you dont find exact number.

msaideroglu, 2023-01-09:
Understood. Thanks.