lnis-uofu/SOFA #144: LEF file for pregenerated eFPGAs

LYWalker, 2023-01-05:
Can you upload a correct lef file for the fpga_top modules? The current one doesn't declare an fpga_top block and has no pin definitions.

This will allow the eFPGAs to be placed and routed using the open-source OpenLane flow.

@ganeshgore 
@yashton 
@GrantBrown1994 

Thank you

ganeshgore, 2023-01-09:
Hello @LYWalker 
Although its completely possible to generate LEF file, 
during our submission for Google MPW1 we used directly GDS file for the integration with Caravel. 
For LEF (with pins) generation for this design I would need sometime. earliest I can address this is second week of February.
If it is urgent you can try to build a LEF from GDS file using MAGIC.

LYWalker, 2023-01-09:
@ganeshgore

Thank you for your reply.

February is good, thanks. 

I'm interested in combining the eFPGA with other components, so if I want to use automated routing I need the lef file with pins.

I've tried using magic to generate the lef file from the gds, but it seems to uncover a magic bug as it gets stuck in an infinite loop - reprinting the same line to the lef file forever.

Thanks

LYWalker, 2023-02-26:
@ganeshgore Hi, any chance this can get generated?