lnis-uofu/SOFA #145: Overlapping macro layouts

mbaykenar, 2023-06-19:
Hi,

When I analyze gds outputs, I see that some macro block layouts are overlapping, such as grid_clb and sb_1__1_:

[image]

Do you have a script/command to avoid of placement of any cell for those overlapping regions? 

Regards,

Burak

ganeshgore, 2023-06-19:
Hello Burak,

The lines you highlighted are just a bounding box of the cell (sb_1__1_). As we used a rectilinear (+) shape, the boundary of the cell and the bounding box of the cell are different. All the standard cells are placed within the boundary of the cell, so the overlapping bounding box is not a problem.

mbaykenar, 2023-06-20:
I didn't notice that. As you mentioned I checked gds of sb_0__1_ and the shapes for switching blocks are not square.

[image]

I assume you chosed these rectiliniear shape to optimize floorplanning of tile-based architecture?

Regards,

ganeshgore, 2023-06-20:
That's right, there are diagonal connections from CLB to SB modules. The rectilinear shape helps in that case.

closed 2023-06-21 by mbaykenar