lnis-uofu/SOFA #147: Cannot run with newest version of OpenFPGA

franktaTian, 2024-04-16:
I use ther docker of  OpenFPGA( according the Document)

Lukemagik, 2024-04-24:
Hi @franktaTian did you solve the problem ?

franktaTian, 2024-05-04:
> Hi @franktaTian did you solve the problem ?

Not yet. I think there maybe some mismatch of version. It is hard for me to debug it.

Lukemagik, 2024-05-04:
Hi @franktaTian If it can be helpful, I solved it by replacing in generate_generate_testbench.openfpga file: '--load_fabric_key ${EXTERNAL_FABRIC_KEY_FILE}' with 'generate_random_key'.

[image]

franktaTian, 2024-05-06:
> Hi @franktaTian If it can be helpful, I solved it by replacing in generate_generate_testbench.openfpga file: '--load_fabric_key ${EXTERNAL_FABRIC_KEY_FILE}' with 'generate_random_key'. [...]

Thank you for your help. I will try it.

franktaTian, 2024-05-07:
Yes,It can work now!

closed 2024-05-07 by franktaTian